phl0/MMDVM_HS_Dual_Hat #2: TCXO COMOS connected to OSC2

sp2ong, 2019-07-03:
Hi 
I have read article RA3APW and he suggests that TCXO CMOS version (in most cases we use) the TCXO should be connected to OSC2 pin put instead OSC1 on ADF7021:

https://www.analog.com/media/en/technical-documentation/data-sheets/ADF7021.pdf

See page 48:

"Set XOSC_ENABLE high when using an external crystal. If using an external oscillator (such as TCXO) with CMOS-level outputs into Pin OSC2, set XOSC_ENABLE low. If using an external oscillator with a 0.8 V p-p clipped sine wave output into Pin OSC1, set XOSC_ENABLE high. "

Source information:
http://www.ra3apw.ru/mmdvm-%D0%B8-tcxo/

phl0, 2019-07-03:
Tnx for information. We will consider that once we make a new revision of the board.

dgiese, 2020-05-02:
Out of curiosity: What effects does the OSC1 instead of OSC2 pin connection have?