phl0/MMDVM_HS_Dual_Hat #6: The STM32F103C8T6's rom size is not enough for USB

jcyfkimi, 2020-06-09:
Hi 
   I just build your board and found that the STM32F103C8T6's ROM size is not enough for USB version, only can go for GPIO. Replacing the MCU with STM32F103CBT6 can make it work. It has 128K ROM.

ra1nb0w, 2021-03-22:
Have you tried the STM32F103CBT6?

jcyfkimi, 2021-03-23:
> Have you tried the STM32F103CBT6?

Yes, I'm using STM32F103CBT6 right now. Works both for GPIO and USB.

ra1nb0w, 2021-03-23:
Ok. Thank you @jcyfkimi

ra1nb0w, 2021-03-23:
Now, the hard point is to find it from a distributor :)

jcyfkimi, 2021-03-31:
Yeah, that's really a disaster. Not sure what's your location, I checked with mouser usa, it's still available, just for the price. Or you can try with aliexpress.

ra1nb0w, 2021-03-31:
In my mouser (EU) it is not available (511-STM32F103CBT6); on order without ETA.

jcyfkimi, 2021-03-31:
Maybe have a try with aliexpress, I know lots of the chips come from aliexpress are recycled and the quailty varies, even with fake chips (for stm32 microcontrollers, I don't think there're fake chips), but considering current stocking status as well as the price issue, I think it's worth to try.

ra1nb0w, 2021-03-31:
Yes, I will try. Thank you.