phl0/MMDVM_HS_Dual_Hat #11: Not working in duplex GPIO

jjsainzc, 2023-09-28:
How can I use as dupkex with pi-star ? I selected MMDVM_HS_Dual_Hat (DB9MAT, DF2ET, & DO7EN) for Pi (GPIO) and not work, only when I use  MMDVM_HS-MDO Hat (BG3MDO) for Pi (GPIO) in simplex mode works, thanks so much, please help me

phl0, 2023-09-29:
> How can I use as dupkex with pi-star ? I selected MMDVM_HS_Dual_Hat (DB9MAT, DF2ET, & DO7EN) for Pi (GPIO) and not work, only when I use MMDVM_HS-MDO Hat (BG3MDO) for Pi (GPIO) in simplex mode works, thanks so much, please help me

Which version of the HAT do you have and which version of firmware pls?

jjsainzc, 2023-09-29:
Sorry my ignorance, how can i get the firmware version of mmdvm hat ?

Best regards
73's HC1LNM
https://www.qrz.com/db/HC1LNM
+593 98 443 0010

Jorge Jesus Sainz Casalla
Freelance senior software developement and consultant.

phl0, 2023-09-29:
> Sorry my ignorance, how can i get the firmware version of mmdvm hat ? Best regards 73's 

It is read upon start of MMDVMHost and written to the logs. What hardware board are you using?

jjsainzc, 2023-09-29:
Hey again

Im using Raspberry Pi 3 Model B Plus Rev 1.3

My radio info show HS_HAT: V1.5.2

Best regards
73's HC1LNM
https://www.qrz.com/db/HC1LNM
+593 98 443 0010

Jorge Jesus Sainz Casalla
Freelance senior software developement and consultant.