phl0/MMDVM_HS_Dual_Hat #13: Enable boot mode on an STM32 aliexpress clone

sysadminpower2019, 2024-05-07:
Hello,

I have been trying to update my firmware for this board but have no had any luck. Apparently it needs to be put into bootloader mode but there does not seem to be any obvious place on the board to work with.

Here is a link to the board. Hopefully at least someone can point me in the right direction to try things out. 

https://www.aliexpress.com/item/1005006281394445.html
[image]

the chip seems to be a geehy apm32 f103cbt6

thanks very much

phl0, 2024-05-07:
This board is not supported by this repo. Please contact the seller. Sorry.

closed 2024-05-07 by phl0

sysadminpower2019, 2024-05-07:
I contacted the seller they said their distributor has no idea either. Any idea where I could turn to support for this board?

phl0, 2024-05-07:
> I contacted the seller they said their distributor has no idea either. Any idea where I could turn to support for this board?

Nope sri.